navaismo/Ender-3V3-SE #121: Octoprint homing grinding

ademharacic3421, 2025-06-05:
### Did you test the latest firmware release?

Yes, and the problem still exists.

### Did you test with the stock Creality firmware?

I have tested, stock firmware has not such problem.

### Bug Description

I home on the display (the lcd) and click the button and it homes like on stock (i changed the homing feedrate and recompiled) and when i home on octoprint, it grinds the motor and makes weird noises.

### Your Slicer

None

### Host Software

None

navaismo, 2025-06-05:
Is this reated to the previous duplicated issue right?
If you recompile the firmware with the values of stock/suggested doesnt matter from where you triggered the homing.

As explainied in other issues this is not a bug, acceleration and feedrate increase generates such sounds. If you think still t noisy you can reduce it until you get the same as stock.

closed 2025-06-05 by navaismo

ademharacic3421, 2025-06-05:
I'm sorry for wasting your time, but i tried recompiling all motor settings and feedrates. It homes nicely but when printing it grinds. 
I have no idea what to do, i really like this firmware because of the input shaping. Could you make a version with stock settings but only add input shaping + linear advance and remove the beeps? That's all.
If not, could you show me how to fix this? The stock one works perfectly fine i have no idea why

navaismo, 2025-06-05:
If this release https://github.com/navaismo/Ender-3V3-SE/releases/tag/v1.0.9.7_9d  still have the sounds then you need to compare thoroughly the changes from Stock to that version(v1.0.9.7_9d). Because the speed and accelaration increasing on 1.0.9.8 version.

eduard-sukharev, 2025-06-05:
¡Hola, Max!
Should we consider reducing the `HOMING_FEEDRATE_MM_M` values? This seem to be recurring issue, we might at least try to reduce the homing speeds - after all, it's only done once per print, or even printer run, not a big deal compared to total print time.

navaismo, 2025-06-05:
Privet @eduard-sukharev! 
Im thinking the same a lot of  people complains because the "Noise", seems they got used to the noises of the stock and the new sounds scare them a lot. Wondering if they go Klipper will ask for the same? Mine is going faster and loder that the released and havent seen issues until i broke it haha.

These are the speeds affected by the changes:
```bash
;   -DHOMING_FEEDRATE_MM_M1=4800,4800,4800
;   -DMAX_ACCEL_EDIT_VALUES1=8000,8000,8000,8000
;   -DDEFAULT_MAX_FEEDRATE1=1000,1000,20,30
;   -DMAX_FEEDRATE_EDIT_VALUES1=10000,10000,40,60
;   -DDEFAULT_MAX_ACCELERATION1=6000,6000,6000,6000

;   -DHOMING_FEEDRATE_MM_M1=300,300,240
;   -DMAX_ACCEL_EDIT_VALUES1=8000,8000,200,8000
;   -DDEFAULT_MAX_FEEDRATE1=500,500,5,30
;   -DMAX_FEEDRATE_EDIT_VALUES1=600,600,10,50
;   -DDEFAULT_MAX_ACCELERATION1=2500,2500,100,2500
```
I'm working on creating `pio` profiles for Fast and Stock speeds with the Baudrates too,  but I'm seeing a big increase of RAM due to that, so now Im going to do a script to generate all FW and people can choose which one.  But not going to be ready until we merge the Bed level branch and the other changes( termistor workaround, remove Mcodes for LCD brigthness, add validation to reste config, etc etc).

ademharacic3421, 2025-06-05:
What could be the cause of this noise that you don't have??? I'm really jealous haha

ademharacic3421, 2025-06-05:
Won't your motors break?? I was scared to death!!! (I'm intermediate-ish with 3d printer knowledge)

eduard-sukharev, 2025-06-05:
It shouldn't affect motors, but it might affect quality and belts lifespan and might induce skipped steps, but you'd want to calibrate for these and do regular maintenance to your printer anyways.

ademharacic3421, 2025-06-05:
Btw, how do you tune input shaping? I have a spare sensor lying around

navaismo, 2025-06-05:
> Won't your motors break?? I was scared to death!!! (I'm intermediate-ish with 3d printer knowledge)

Nope, listen:

https://github.com/user-attachments/assets/b5b1bf3a-f513-4878-a5f6-11c392e10d04

> Btw, how do you tune input shaping? I have a spare sensor lying around

If you are suing octoprint you can use: https://github.com/navaismo/Octoprint-Pinput_Shaping
If not, print the ringing tower: https://marlinfw.org/docs/gcode/M593.html

ademharacic3421, 2025-06-06:
Btw does it affect print quality? And also when i open octoprint with a server on octo4a, i have baud rate 115200, and i try to upload infos of the print thing and the picture of file, the layer and etc, it overlaps my main menu screen and uploads reaallly slowly like 11-12kbps-ish

eduard-sukharev, 2025-06-06:
> i have baud rate 115200, and i try to upload infos of the print thing and the picture of file, the layer and etc, it overlaps my main menu screen and uploads reaallly slowly like 11-12kbps-ish

That's exactly what it is — baudrate is literally baud/s, where baud is characters per second, the actual figures are bits/s, which means our transmission rate is 115200 bits/s = 14400 bytes/s = 14 kibibytes/s. Factor in some communication protocol overhead and there you have it — 11-12 kbps.