navaismo/Ender-3V3-SE #144: No serial port found

Stixz, 2025-07-10:
### Did you test the latest firmware release?

Yes, and the problem still exists.

### Did you test with the stock Creality firmware?

I have tested, stock firmware has not such problem.

### Bug Description

I installed the OctoPrint for the 115200 baud and followed the step for the settings and the plugins, and I am unable to get OctoPrint to see/connect to the Port. I have also checked and triple checked my cables,

Any assistance you can offer here, would be wonderful!

Thanks!

### Your Slicer

None

### Host Software

None

TheDarkPreacher, 2025-07-11:
Having the same issue. Using FW10981f_115Kbps_Normal, version 0.2.9 of the plugin. Attached are the screenshot and my logs.

[image]

[octoprint-logs.zip](https://github.com/user-attachments/files/21189443/octoprint-logs.zip)

navaismo, 2025-07-11:
> Having the same issue. Using FW10981f_115Kbps_Normal, version 0.2.9 of the plugin. Attached are the screenshot and my logs.

I Just flashed and tried that version is working good on my side:

```diff
2025-07-11 15:02:33,757 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Detecting serial connection"
2025-07-11 15:02:33,832 - octoprint.server - INFO - Autorefresh of serial port list stopped
2025-07-11 15:02:33,884 - octoprint.util.comm - INFO - Serial detection: Performing autodetection with 2 port/baudrate candidates: /dev/ttyUSB0@115200, /dev/ttyS0@115200
+2025-07-11 15:02:33,885 - octoprint.util.comm - INFO - Serial detection: Trying port /dev/ttyUSB0, baudrate 115200
+2025-07-11 15:02:33,921 - octoprint.util.comm - INFO - Connecting to port /dev/ttyUSB0, baudrate 115200
+2025-07-11 15:02:34,043 - octoprint.util.comm - INFO - Serial detection: Handshake attempt #1 with timeout 5.0s
+2025-07-11 15:02:34,065 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
+2025-07-11 15:02:34,148 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial connection" to "Operational"
2025-07-11 15:02:34,160 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
+2025-07-11 15:02:34,317 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin V1.0.9.8_1f (Jun 25 2025 17:44:37)"
2025-07-11 15:02:34,317 - octoprint.util.comm - INFO - Firmware info line: FIRMWARE_NAME:Marlin V1.0.9.8_1f (Jun 25 2025 17:44:37) SOURCE_CODE_URL:https://github.com/navaismo/Ender-3V3-SE/tree/main PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender-3 V3 SE EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
2025-07-11 15:02:34,452 - octoprint.util.comm - INFO - Firmware states that it supports temperature autoreporting
2025-07-11 15:02:34,764 - octoprint.util.comm - INFO - Firmware states that it supports emergency GCODEs to be sent without waiting for an acknowledgement first
2025-07-11 15:02:35,582 - octoprint.util.comm - INFO - Firmware sent the following capability report:
  SERIAL_XON_XOFF: not supported
  BINARY_FILE_TRANSFER: not supported
  EEPROM: supported
  VOLUMETRIC: supported
  AUTOREPORT_POS: not supported
  AUTOREPORT_TEMP: supported
  PROGRESS: not supported
  PRINT_JOB: supported
  AUTOLEVEL: supported
  RUNOUT: supported
  Z_PROBE: supported
  LEVELING_DATA: supported
  BUILD_PERCENT: supported
  SOFTWARE_POWER: not supported
  TOGGLE_LIGHTS: not supported
  CASE_LIGHT_BRIGHTNESS: not supported
  EMERGENCY_PARSER: supported
  HOST_ACTION_COMMANDS: supported
  PROMPT_SUPPORT: supported
  SDCARD: supported
  REPEAT: not supported
  SD_WRITE: supported
  AUTOREPORT_SD_STATUS: not supported
  LONG_FILENAME: supported
  THERMAL_PROTECTION: supported
  MOTION_MODES: not supported
  ARCS: supported
  BABYSTEPPING: supported
  CHAMBER_TEMPERATURE: not supported
  COOLER_TEMPERATURE: not supported
  MEATPACK: not supported
```

[FW10981f_115Kbps_Normal.zip](https://github.com/user-attachments/files/21189562/FW10981f_115Kbps_Normal.zip)

navaismo, 2025-07-11:
> Firmware Ver. FW10981ePIC_115Kbps_Normal

Try the 128K version maybe:

[fw981ePic.bin.zip](https://github.com/user-attachments/files/21189569/fw981ePic.bin.zip)

```
+2025-07-11 15:07:50,136 - octoprint.util.comm - INFO - Serial detection: Performing autodetection with 2 port/baudrate candidates: /dev/ttyUSB0@128000, /dev/ttyS0@128000
+2025-07-11 15:07:50,137 - octoprint.util.comm - INFO - Serial detection: Trying port /dev/ttyUSB0, baudrate 128000
2025-07-11 15:07:50,145 - octoprint.util.comm - INFO - Connecting to port /dev/ttyUSB0, baudrate 128000
2025-07-11 15:07:50,163 - octoprint.util.comm - INFO - Serial detection: Handshake attempt #1 with timeout 5.0s
2025-07-11 15:07:50,173 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
+2025-07-11 15:07:50,197 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial connection" to "Operational"
2025-07-11 15:07:50,209 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
+2025-07-11 15:07:50,259 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin V1.0.9.8_1ePIC (Jun 20 2025 20:06:56)"
2025-07-11 15:07:50,259 - octoprint.util.comm - INFO - Firmware info line: FIRMWARE_NAME:Marlin V1.0.9.8_1ePIC (Jun 20 2025 20:06:56) SOURCE_CODE_URL:https://github.com/navaismo/Ender-3V3-SE/tree/main PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender-3 V3 SE EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
2025-07-11 15:07:50,360 - octoprint.util.comm - INFO - Firmware states that it supports temperature autoreporting
2025-07-11 15:07:50,529 - octoprint.server - INFO - Autorefresh of serial port list stopped
2025-07-11 15:07:50,606 - octoprint.util.comm - INFO - Firmware states that it supports emergency GCODEs to be sent without waiting for an acknowledgement first
2025-07-11 15:07:51,559 - octoprint.util.comm - INFO - Firmware sent the following capability report:
  SERIAL_XON_XOFF: not supported
  BINARY_FILE_TRANSFER: not supported
  EEPROM: supported
  VOLUMETRIC: supported
  AUTOREPORT_POS: not supported
  AUTOREPORT_TEMP: supported
  PROGRESS: not supported
  PRINT_JOB: supported
  AUTOLEVEL: supported
  RUNOUT: supported
  Z_PROBE: supported
  LEVELING_DATA: supported
  BUILD_PERCENT: supported
  SOFTWARE_POWER: not supported
  TOGGLE_LIGHTS: not supported
  CASE_LIGHT_BRIGHTNESS: not supported
  EMERGENCY_PARSER: supported
  HOST_ACTION_COMMANDS: supported
  PROMPT_SUPPORT: supported
  SDCARD: supported
  REPEAT: not supported
  SD_WRITE: supported
  AUTOREPORT_SD_STATUS: not supported
  LONG_FILENAME: supported
  THERMAL_PROTECTION: supported
  MOTION_MODES: not supported
  ARCS: supported
  BABYSTEPPING: supported
  CHAMBER_TEMPERATURE: not supported
  COOLER_TEMPERATURE: not supported
  MEATPACK: not supported
```

TheDarkPreacher, 2025-07-11:
Downloaded the [FW10981f_115Kbps_Normal.zip](https://github.com/user-attachments/files/21189562/FW10981f_115Kbps_Normal.zip) that you linked, installed, rebooted everything, and now I can connect. One of two things possibly happened: It didn't like the rename I used when I renamed the firmware something random, or the original file I downloaded was wrong, either because of a hiccup in my connection, or you have a slightly bugged version uploaded that you fixed later. Who can say? Either way, my issue is fixed, and I'm prepping a test print. Thanks, navaismo!