navaismo/Ender-3V3-SE #154: Adaptive Bed Mesh Leveling

Sidiusz, 2025-07-23:
### Describe it

Hi! Is it possible to add `G29 L R F B` support to the firmware? Thanks!

navaismo, 2025-07-23:
I think is already present have you tried it?
If so whats the output and the behaviour is not working as expected?

navaismo, 2025-07-23:
> oh, one more thing, so as not to create a new issue
> Is this a known problem, or is it just me that the sdcard is unreadable until I turn off and turn on the printer with it inserted? It's like it's empty, and the screen "lags" when I insert or pull it out, as if it's trying to read or update something

I don't use the SD since I installed octoprint last year so never noticed that, not sure if is somthing new or that happened long before.

Need to finish some prints and later I can check both on my printer and get back to you regarding the G29 and the SD Card.
Your Machine has C13 or C14 hardware?

Sidiusz, 2025-07-23:
C14

navaismo, 2025-07-24:
Regarding the SDCARD issue I was able to replicate it only when Octoprint was Connected a.k.a Serial Active Connection enable.

When I didn't connect octoprint SD Print works as expected. So why when I enable the Serial Connection it doesn't work?

Well the answer lays in the way I made the PAUSE|START/RESUME options to have a better control when using the HOST(Octoprint):

```c++
/**
  * M24: Start or Resume SD Print
  */
 void GcodeSuite::M24()
 {
 
    SERIAL_ECHOLNPAIR("=====++++>> M24 Resume & Flag = ", serial_connection_active);
   if (serial_connection_active)
   {
      SERIAL_ECHOLN("=====++++>> M24 with Octo, Sending Resume Command");
 #if ENABLED(HOST_ACTION_COMMANDS)
 #ifdef ACTION_ON_RESUME
     host_action_resume();
 #endif
     TERN_(HOST_PROMPT_SUPPORT, host_prompt_open(PROMPT_INFO, PSTR("Resuming"), DISMISS_STR));
 #endif
   }
   else
...
```

When the serial connection is active the instructions are sent to the HOST, so the Start|Resume option trigger Octoprint's M27 Status, and sice the printer technically is not printing the job stuck in an endless loop trying to find when is going to start.

### How to solve this?

You have two options:

1. If you are planning to use the SDCARD, do not connect Octoprint no need if the control is on the Printer's side/LCD Knob.
2. If for some reason you really really need to connect Octoprint but trigger the jobs from SDCARD, you can disable the Serial flag by sending from Octoprint's terminal the command:

```bash
O9000 OCOFF|
```
The command will disable the priority HOSTS commands and leverage all the logic to the printer and will disable the features of the E3V3SEPrintJobDetails Plugin until the flag is enable again.  

I'm still looking at the  G29 limits and let you know later.

Sidiusz, 2025-07-25:
И зачем я парился с английским...

eduard-sukharev, 2025-07-25:
I did sometimes get an unreadable SD card and I'm not sure what was the reason, I thought it was due to incorrect sd card eject from the PC (like, without prior unmounting) and at that point the sd card becomes unreadable and I had to use another sd card. But once in a while the first one would become readable again so I was assuming that was just cheap SD cards and that's it. Never tried to "fix" that with printer reboot with sd card being inserted.

Anyways, not sure how and it should be solved in firmware.

Sidiusz, 2025-07-26:
> I did sometimes get an unreadable SD card and I'm not sure what was the reason, I thought it was due to incorrect sd card eject from the PC (like, without prior unmounting) and at that point the sd [...]

Спасибо аз ответ. Это точно не связано с неправильным извлечением карты из ПК и не с "дешёвой" картой потому, что две разные карты от двух разных производителей в разном ценовом сегменте ведут себя абсолютно идентично, на официальной прошивке от creality такого никогда не происходило, вообще ни разу

Sidiusz, 2025-07-26:
okay, cancel the panic
as a friend told me (this is his printer), the problem with the sd card started after installing firmware 1.0.7 (also custom from the community)
It's strange that your version didn't completely replace it (although I installed 1.0.6 from the creality website on both the printer and the screen)

closed 2025-08-04 by navaismo