navaismo/Ender-3V3-SE #156: Live skew correction

M2rsh, 2025-07-25:
### Describe it

Would there be a way to enable marlin skew correction while also being able to change its settings without recompiling the software?

navaismo, 2025-07-25:
The skew compensation can be calculated manually according the `Configuration.h` file:

 ```c++
/**
 * Bed Skew Compensation
 *
 * This feature corrects for misalignment in the XYZ axes.
 *
 * Take the following steps to get the bed skew in the XY plane:
 *  1. Print a test square (e.g., https://www.thingiverse.com/thing:2563185)
 *  2. For XY_DIAG_AC measure the diagonal A to C
 *  3. For XY_DIAG_BD measure the diagonal B to D
 *  4. For XY_SIDE_AD measure the edge A to D
 *
 * Marlin automatically computes skew factors from these measurements.
 * Skew factors may also be computed and set manually:
 *
 *  - Compute AB     : SQRT(2*AC*AC+2*BD*BD-4*AD*AD)/2
 *  - XY_SKEW_FACTOR : TAN(PI/2-ACOS((AC*AC-AB*AB-AD*AD)/(2*AB*AD)))
 *
 * If desired, follow the same procedure for XZ and YZ.
 * Use these diagrams for reference:
 *
 *    Y                     Z                     Z
 *    ^     B-------C       ^     B-------C       ^     B-------C
 *    |    /       /        |    /       /        |    /       /
 *    |   /       /         |   /       /         |   /       /
 *    |  A-------D          |  A-------D          |  A-------D
 *    +-------------->X     +-------------->X     +-------------->Y
 *     XY_SKEW_FACTOR        XZ_SKEW_FACTOR        YZ_SKEW_FACTOR
 */
```

So after calculating  the Hypotenuse (AB) with the above formula, then you can calculate the XY Factor and set it using the [M582](https://marlinfw.org/docs/gcode/M852.html) Command.

But if somehow you prefer to edit the values of:

```c++
   #define XY_DIAG_AC 282.8427124746
  #define XY_DIAG_BD 282.8427124746
  #define XY_SIDE_AD 200
```

You can create a global Variables that can be edited and maybe (not necessarily) stored in the EEPROM, then you can just assign a value to those variables using a Custom LCD Menu or Custom Gcode.

TooMuchAndNotEnough, 2025-07-27:
I actually came here to check for skew correction too. I don't think skew correction and M852 is enabled in the configuration.h file though? I've never worked in the Marlin code base before, but I believe the file I linked below is where we would make that change:

https://github.com/navaismo/Ender-3V3-SE/blob/99c5f4f18a1ad4856d85a82321bad231b63d2b7a/Marlin/Configuration.h#L1768

The work you guys have done here is awesome though, this repo adds so much functionality, thank you for making this!

Dan263737, 2025-07-30:
good idea

eduard-sukharev, 2025-07-31:
> Edit: I changed that line in the config and the one to enable the M852 instruction. Now I'm getting skew correction with just the M852 instruction, cool!

When you built your firmware, the terminal should've reported RAM and Flash usage. Can you post it here? It might've raised RAM usage and that might lead to printer halt and reboot during the print at random moments.

navaismo, 2025-11-23:
We created the menus and the logic but is consuming a lot of Memory affecting the print jobs so for the next release the MCODE will be enabled so you can set it from Serial port.

closed 2025-11-23 by navaismo