navaismo/Ender-3V3-SE #160: Print Complete Dialogue Appears Incorrectly

hasanAUST, 2025-08-14:
### Did you test the latest firmware release?

Yes, and the problem still exists.

### Did you test with the stock Creality firmware?

I have tested, stock firmware has not such problem.

### Bug Description

I have noticed 3 issue together. 
1. Sometimes the display shows remaining print time > 100H ( I have read that it's happening for small files, so i ignored it)
2. Print complete dialogue appears before the actual print is yet to complete.
3. After print complete dialogue appears, print percentage goes backwards, e.g.: 100 to 98, 95 etc. and then goes up again!

https://github.com/user-attachments/assets/e4ae2d31-63a5-4fb3-88b3-1e5cbcb3f238

Point 2 and 3 happening regularly.
Although, none of these affecting print quality. 

### Expected behavior

Print Complete Dialogue supposed to appear after actual print is complete.

After print complete, it should show 100 percent. 

### Actual behavior

Print Complete Dialogue appear before actual print is complete.

After print complete, print progress number goes down and goes upto a number (not 100).

### Steps to Reproduce

1. Slice a model
2. Load the gcode to SD card
3. Print the file

### Your Slicer

Orca Slicer

### Host Software

SD Card

### Additional information & file uploads

Gcode File

[200c gcode.txt](https://github.com/user-attachments/files/21780067/200c.gcode.txt)

navaismo, 2025-08-14:
> 1. Sometimes the display shows remaining print time > 100H ( I have read that it's happening for small files, so i ignored it)

Already explained in duplicated threads, this is due to how the printer reads the file, the calculation is based on the progress of the file which came from stock routines.

> 2. Print complete dialogue appears before the actual print is yet to complete.

Also answered in other threads, please review the closed issues. You dont have real time updates because it affects your prin quality so the Stock routine to show everything is polling every N milliseconds.

>3.  After print complete dialogue appears, print percentage goes backwards, e.g.: 100 to 98, 95 etc. and then goes up again!

@eduard-sukharev this could be an issue on the SD routine to read the progress? I havent touched the SD progress code, by any chance do you identify an issue here?

>Steps to Reproduce
Slice a model
Load the gcode to SD card
Print the file

I haven use the SD since looong time ago, because I focus on Octoprint so can´t tell about if this was present in stock or not(and now my printer is disassembled so cant test either). Let's see if Eduard has an insight on this to work with.

Dan263737, 2025-08-15:
mine does this too

eduard-sukharev, 2025-08-15:
That's an interesting behavior. I think the reason is that we show the finish screen a little too early and also continue to update progress even after it reached 100%.

We should probably fix both, but fixing either should be fine.

navaismo, 2025-08-21:
> I dont have this isuue

It is only related with SD print. If you are using Octoprint, no we dont have it.