navaismo/Ender-3V3-SE #163: Ender 3 V3 SE sometimes hangs after starting a new print or shows incomplete calibration

kingchenc, 2025-08-22:
### Did you test the latest firmware release?

Yes, and the problem still exists.

### Did you test with the stock Creality firmware?

I have tested, stock firmware has not such problem.

### Bug Description

I’ve noticed a couple of weird issues with my Ender 3 V3 SE running the latest community firmware:

	1.	Random hang after print: About 20-40% of the time (seems totally random), when a print finishes and I start a new one, the printer moves to the middle to start calibrating/starting… and then just hangs there. Only power off/on fixes it. Values on the Display shows only 0!
	

	2.	Calibration weirdness: Today it happened again. I started a new print after hanging. Normally, the printer does a 7x7 calibration grid (always activated), but today it only did the 4 outer squares. First time i see that

### Steps to Reproduce

I cant, its totaly random :/

### Your Slicer

Orca Slicer

### Host Software

SD Card

Dan263737, 2025-08-22:
Have similar problem

Dan263737, 2025-08-24:
does anybody even read this?

navaismo, 2025-09-18:
> slowed down...

You mean feedrate slowed down(print speed)?
That's totally random never seen and can't think why.

FYI I already started to work in a new version using MeatPack (if people use the serial port) to optimize communication and also I want to rewrite the Octoprint Plugin to reduce to stock values the Serial Buffers which affects a lot the usage of the memory.

kingchenc, 2025-09-18:
> > slowed down...
> 
> You mean feedrate slowed down(print speed)? That's totally random never seen and can't think why.
> 
> ...

If you mean with "feedrate" the travel and print movement speed then yes.

kingchenc, 2025-09-19:
> Yep, this part sets the values from slicer:
> 
> ; EXECUTABLE_BLOCK_START
> M201 X1000 Y500 Z250 E1000
> M203 X250 Y250 Z5 E40
> M204 P2500 R500 T250 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
> M205 X5.00 Y4.00 Z0.30 E4.00 ; sets the jerk limits, mm/sec

So, what does that mean now? I just have very "slow" settings, because my Ender is sitting next to my PC ^^ 

But why does this only happen very rarely? Sometimes after another print, or like today when I started it after 2 days offline? 

How can this be "fixed"? By myself or by the firmware?

kingchenc, 2025-09-19:
> Propagation is not firmware related is your slicer doing it through the Gcode file, disable the feature or match the speeds and acceleration in both(printer & Slicer)

Should I adjust the speed and acceleration settings on my Ender to match what the slicer is generating, or is it better to disable the slicer from writing speed/acceleration commands into the G-code? 

Im not sure which way is the proper fix. 

But what confuses me: why did this ‘bug’ not appear for almost half a year with the same settings, and now suddenly it does?

navaismo, 2025-09-19:
> Should I adjust the speed and acceleration settings on my Ender to match what the slicer is generating, or is it better to disable the slicer from writing speed/acceleration commands into the G-code?
> 
> Im not sure which way is the proper fix.

Whatever works best for you. In my case I disable it.

 

> But what confuses me: why did this ‘bug’ not appear for almost half a year with the same settings, and now suddenly it does?

No idea, every setup has its own settings and external variables, can't tell if is a firmware because never happened to other nor me or slicer update related issue/feature.