navaismo/Ender-3V3-SE #166: [DRAFT] Extruder trying to overextrude like crazy

Jcodeerd, 2025-09-07:
### Did you test the latest firmware release?

Yes, and the problem still exists.

### Did you test with the stock Creality firmware?

No, but I will test it now!

### Bug Description

When trying to print this file, the extruder suddenly starts extruding a ton of filament, resulting in big blobs, and breaking the filament.

### Expected behavior

The print to print successfully

### Actual behavior

The printer started overextruding 

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

https://github.com/user-attachments/assets/f5609d8b-40e3-49cd-8416-77a73366bde8
Running v1.0.9.8_1ePIC with GCODE preview

navaismo, 2025-09-07:
Which Gcode file?

Jcodeerd, 2025-09-07:
https://i.kaashout.nl/Overextruding.gcode

Note that I haven't tried with the stock firmware yet, so I'm not 100% sure whether its this custom firmware being the issue

Jcodeerd, 2025-09-08:
I haven't taken a photo of the buildplate specifically, but you can see a bit in the video. It's supposed to be like this: 

[image]

navaismo, 2025-09-14:
> https://i.kaashout.nl/Overextruding.gcode
> 
> Note that I haven't tried with the stock firmware yet, so I'm not 100% sure whether its this custom firmware being the issue

I've tried your gcode, no issues. Same FW v1.0.9.8_1ePIC. Could be a clog, clean the nozzle and trya again.

[image]

https://github.com/user-attachments/assets/a93b1dbc-ea28-4834-89e9-35a67b6ae9ab