navaismo/Ender-3V3-SE #169: LCD stay in the home page when octoprint start print job

diegoemy76, 2025-09-14:
### Did you test the latest firmware release?

Yes, and the problem still exists.

### Did you test with the stock Creality firmware?

I have tested, stock firmware has not such problem.

### Bug Description

when I send the print job from octoprint the printer lcd seams like it did not received any data. the printer print but the lcd stay in the home page Saying octoprint connected  The Pi is connected via serial directly from the printer not via USB exactly as it was before with esp32 but with esp3d if I sent the print command via web Page the display showed the information About the print job 

### Your Slicer

None

### Host Software

None

diegoemy76, 2025-09-14:
[image]

It remains like this without displaying anything about the print

navaismo, 2025-09-14:
Does the plugin is properly configured?

Follow the readme and clarify the connection, firmware version and speed used.
https://github.com/navaismo/OctoPrint-E3v3seprintjobdetails?tab=readme-ov-file#configure-the-plugin

navaismo, 2025-09-14:
Of course, Plugin is meant to use with the firmware + Octoprint. If you dont use the plugin & Octoprint then you will have default LCD behaviour.

closed 2025-09-14 by diegoemy76